mayeranalytics/pySX127x #6: Weird behaviour using the beacon class for some bytes

thomasverbeke, 2016-02-01:
When sending some byte for example "0x00", or "0x67" the interrupt (TxDone) will not be generated the third time. This is a very weird bug? It can be reproduced by changing the 1-byte message (0x0f) to 0x00 or 0x67 for example in the beacon. When extending the message to multiple bytes the same thing will happen when these bytes are part of the larger message.

mayeranalytics, 2016-03-28:
Very weird.
What board/chip are you using?

thomasverbeke, 2016-03-28:
This happened to me a long time ago and if I remember well it was related to a bug in your code which I fixed in my fork. I will review my code tomorrow to see how I solved it.

Abrasam, 2016-07-14:
I'm having this issue with the HopeRF FRM95W.

bjcarne, 2016-08-08:
I would also be interested to know how to resolve this. Any payload I send that ends with binary 000 in the last byte (e.g. hex 0xf[0-7]) causes the tx_done to not be generated on the next transmission.

It doesn't seem to matter what any of the other bytes contain, just the last one in the payload is affected.

I'm using Raspberry Pi 2 and inair9b.

bjcarne, 2016-08-09:
I have found the fix in https://github.com/thomasverbeke/pySX127x/issues/1#issuecomment-178582087. It is in relation to the clearing of IRQ flags.

Pull request - https://github.com/mayeranalytics/pySX127x/pull/16

I have tested with Raspberry Pi 2 and inAir9B using Python 3.4.2.

closed 2017-06-08 by mayeranalytics