mayeranalytics/pySX127x #7: AssertionError running lora_util.py

KentuckyFriedFrank, 2016-03-28:
Hi,
I am receiving the following after following the Instruction section and running 'sudo ./lora_util.py'

> /home/pi/projects/pySX127x/SX127x/board_config.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(BOARD.LED, GPIO.OUT)
Traceback (most recent call last):
  File "./lora_util.py", line 35, in <module>
    lora = LoRa(verbose=False)
  File "/home/pi/projects/pySX127x/SX127x/LoRa.py", line 92, in __init__
    self.rx_chain_calibration(968.)
  File "/home/pi/projects/pySX127x/SX127x/LoRa.py", line 835, in rx_chain_calibration
    self.set_freq(freq_bkup)
  File "/home/pi/projects/pySX127x/SX127x/LoRa.py", line 271, in set_freq
    assert self.mode == MODE.SLEEP or self.mode == MODE.STDBY or self.mode == MODE.FSK_STDBY
AssertionError

I am running this on a Raspberry Pi 2 Model B with the latest RASPBIAN JESSIE and a InAir4 433Mhz LoRa module.

Please let me know if can provide any more information.

Thanks

thomasverbeke, 2016-03-28:
Reset your LoRa board or Raspberry Pi.

dmahony, 2016-03-28:
I had a similar issue when my wireless tranceiver wasnt connected properly.

KentuckyFriedFrank, 2016-03-28:
@thomasverbeke to be clear, this can be done by power cycling the LoRa board or Raspberry Pi, correct?

@dmahony  can you please explain in your case what was wrong with the soldering? 

Thank you both for you input.

thomasverbeke, 2016-03-28:
this is an assertion error; so your device is not in SLEEP/STDBY or FSK_STDBY mode.
you can manually poll the register (look at my fork for examples) to check what mode your device is in; if you have a working spi implementation this should be very easy to test.

I would occasionally get this error when my lora device was already in continuous reception or tx mode; then the assertion would fail and the only way to quickly solve this problem is to reset my board (since I included an easy reset button on my pcb).

You could also get this error of course If you have bad connections for the SPI communication lines (MOSI/MISO/CLK/Slave select) due to bad soldering and I think this is what dmahony  is refering to.

Good luck!

KentuckyFriedFrank, 2016-03-28:
Thanks @thomasverbeke for the response. Right now I am going to trace down my soldering job. Looking over it again it doesn't look so good. I will let you know how that goes.

Thanks again!

mayeranalytics, 2016-03-28:
@KentuckyFriedFrank The pinout looks ok to me. You have the inAir4 with SX11276 or SX1278. The Medtronix specs say: "This module can be assembled with the SX1276 or SX1278 chip. Both chips are identical when used for 433MHz operation." 
But the frequency ranges do indeed differ:
SX1276: 137 - 1020 MHz
SX1278: 137 - 525 MHz
So maybe you're on the right track with your suspicion that you shouldn't change to 868MHz.

I don't yet see how this would explain the assertion failure, though - the assert is called before the frequency change is attempted. But I'll have a closer look.

In any case it's not a good idea to switch to 868MHz when you have the 433MHz circuitry and antenna.
I added an additional parameter to LoRa.__init__ that lets you pick another frequency for calibration than 868MHz. (The default is still 868MHz, though!)

dmahony, 2016-03-28:
My tranceiver was connected correctly, Its just the soldering needed redoing then it was fine.

KentuckyFriedFrank, 2016-03-28:
@mayeranalytics  thanks for the follow up! I see in LoRa.py Line 92 the changes you made: 968 >> calibration_freq (which calibration_freq is set to 868 on line 84). I am just curious why the change from 968 to 868.

KentuckyFriedFrank, 2016-03-29:
After re-soldering all solder joints I am still getting the same issue. Even with the new code and setting calibration_freq = 433.

I will let you know if I make any progress.

KentuckyFriedFrank, 2016-03-29:
Ok, I have no clue what is wrong. 
I thought it might be the RPI SPI so I tried to find a RPI to Arduino SPI test.
I cannot even get the RPI SPI to Arduino to work using this tutorial:
http://mitchtech.net/raspberry-pi-arduino-spi/ (no HELLO WORLD)
even though I have ran the RPI SPI LOOPBACK TEST successfully.
https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md

KentuckyFriedFrank, 2016-03-29:
I really do not know what is wrong. I have verified that SPI is working using communication via SPI with an Arduino http://mitchtech.net/raspberry-pi-arduino-spi/ as well as the LOOPBACK TEST https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md. I have also verified the pins are good and making a connection to the RPI pins. I have also swapped out the InAir4 module with 2 others and have the same problem.

KentuckyFriedFrank, 2016-03-29:
> Does CS need to be attached to a GPIO? 
This was what was causing the issues.
If only there was a Head-to-Wall emoticon on GitHub... 

My final pinout:
inAir4 DIO0 >> GPIO 22(pin 15)
inAir4 DIO1 >> GPIO 23(pin 16) 
inAir4 DIO2 >> GPIO 24(pin 18) 
inAir4 DIO3 >> GPIO 25(pin 22) 
inAir4 Reset>> NA
LED >> GPIO 18(pin 12)
inAir4 VS >> GPIO 04(pin 07)
inAir4 3V3 >> 3.3v(pin 01)
inAir4 V0 >> Ground(pin 09)
inAir4 CK >> GPIO 11 (pin 23)
inAir4 SO >> GPIO 09 (pin 21)
inAir4 SI >> GPIO 10 (pin 19)
inAir4 CS >> GPIO 08 (pin 24)

Thanks for your help everyone!

closed 2016-03-29 by KentuckyFriedFrank

mayeranalytics, 2016-03-29:
@ KentuckyFriedFrank 968 is a typo. 968MHz is not in any ISM band.