mayeranalytics/pySX127x #11: connecting to Pi

aabognah, 2016-06-15:
Hi,

This is more of a request than an issue.

Can you post a picture or a diagram of the connections between the Pi and inAir9.

Thanks.

mayeranalytics, 2016-07-03:
In the README there is a table that shows how a Raspberry rev 2 is connected to the inAir9B.

KentuckyFriedFrank, 2016-08-02:
Hi Aabognah,

You can also refer to my pinout (using InAir4) in https://github.com/mayeranalytics/pySX127x/issues/7 

> My final pinout:
> inAir4 DIO0 >> GPIO 22(pin 15)
> inAir4 DIO1 >> GPIO 23(pin 16) 
> inAir4 DIO2 >> GPIO 24(pin 18) 
> inAir4 DIO3 >> GPIO 25(pin 22) 
> inAir4 Reset>> NA
> LED >> GPIO 18(pin 12)
> inAir4 VS >> GPIO 04(pin 07)
> inAir4 3V3 >> 3.3v(pin 01)
> inAir4 V0 >> Ground(pin 09)
> inAir4 CK >> GPIO 11 (pin 23)
> inAir4 SO >> GPIO 09 (pin 21)
> inAir4 SI >> GPIO 10 (pin 19)
> inAir4 CS >> GPIO 08 (pin 24)

closed 2017-06-08 by mayeranalytics