mayeranalytics/pySX127x #18: More of a question, since I didn't see where to post this here...

gregarican, 2017-03-04:
I have a Cypress USB-UART that has the SX1272 chipset. Looking to use this on my Windows 7 Pro 64-bit laptop. As a receiver, since the transmitter is an Arduino Uno with a Dragino Lora GPS shield on top of it. The transmitter is working fine. Would like to avoid setting up another Arduino for the receiver, since my laptop already has WinPython on it.

Will your Python library work under Windows?

dziedzic99, 2017-03-04:
I guess you should go with Arduino nano for the receiver as it natively
supports SPI. I have tested this solution and it worked.

mayeranalytics, 2017-03-05:
The library is written in python so it certainly will run on Windows. What you have to figure out is how to communication with the chip (read/write the registers). What Cypress kit do you use?

gregarican, 2017-03-06:
Actually I found a Python library on the hardware vendor's (iFrogLab) website. With some example programs. It appears to be supported under Windows, mapping a Windows COM port to the LoRA device. Will test out later and confirm the functionality exists. I realize that Python should be compatible with Windows, Mac, Linux, etc. but it isn't always 100% true. Depending on how in-depth the Python libraries are in terms of the underlying OS. And I also realize I could resort to Cygwin or something if it wasn't truly native Windows compatible.

Since the Windows PC will be a Lora receiver --- taking transmission telemetry, environmental measurements, and chunked image files --- I need to have the LoRA listener perform some heavy lifting. Much for comfortable developing that on a true workstation, rather than let Arduino handle it in a more disconnected fashion in terms of my workstation!

closed 2017-03-06 by gregarican

gregarican, 2017-03-07:
Sorry, had one last question :) Since I am still trying to communicate with this (http://www.ifroglab.com/en/?p=6536) using a Windows-based USB UART connection, what's the quickest way to implement a small test set? Your project's test_lora.py fails, since obviously it's looking for RPi module pins. Basically I would like to communicate with the SX1272 via UART instead. Any quick suggestions?

mayeranalytics, 2017-03-07:
In a nutshell, replace every SpiDev call (spi.xfer) in SX127x/LoRa.py, and put in the serial communication via UART. And remove the BOARD class.
There are many spi.xfers but they can be replaced in one go.