mayeranalytics/pySX127x #19: AssertionError

TNAnouar, 2017-03-16:
Hi, 
I m trying to use your libraray so i can use the sx1276. 
I got an error while trying tot run the file  lora_util.py even i tried with version3 of python.
the error is he following:

Traceback (most recent call last):
  File "lora_util.py", line 35, in <module>
    lora = LoRa(verbose=False)
  File "/home/pi/Desktop/pySX127x/SX127x/LoRa.py", line 100, in __init__
    self.rx_chain_calibration(calibration_freq)
  File "/home/pi/Desktop/pySX127x/SX127x/LoRa.py", line 856, in rx_chain_calibra
    self.set_freq(freq_bkup)
  File "/home/pi/Desktop/pySX127x/SX127x/LoRa.py", line 282, in set_freq
    assert self.mode == MODE.SLEEP or self.mode == MODE.STDBY or self.mode == MO
AssertionError

Can you help me please..??

mayeranalytics, 2017-03-17:
This error means that you (i.e. lora_util.py) tried to set the frequency while being in the wrong mode for that action. lora_util.py should certainly work when the SX1276 is in a default state. Can you reset the chip (get it into default state, e.g. by powering off and on) and try again?

TNAnouar, 2017-03-17:
Hi,
Thank you for your quick answer ,I will try like you said and I m gone give
you the result

radioman85, 2017-04-04:
My problem was (receiving the same message) that I didn't connected the pins properly!! (forgot to connect the pin 'SI')

closed 2017-06-08 by mayeranalytics