mayeranalytics/pySX127x #21: Use with Dragino Lora GPS Hat

phraktle, 2017-07-12:
Hi,

I'm trying to use the library with a Dragino Lora GPS Hat v1.4 board on a RPi3, with no luck so far (assertion error in set_freq during calibration). Has anyone successfully used this board? With what settings?

Thanks,
  Viktor

pointhi, 2017-09-03:
You need to do some little changes to the board configuration and on the board itself:

* in the board class: ```DIO0 = 4```
* solder a jumper between pin 22 and 24 as done in the [LowCostLoRaGw tutorial](http://wiki.dragino.com/index.php?title=Lora/GPS_HAT#Example4_--_Set_up_as_a_LoRa_Gateway_using_the_LowCostLoRaGw_library). The background is that Dragino didn't used the normal chip select pin which are supported by spidev, which requires this dirty hack (or some software workaround)

acceleratiom, 2017-12-06:
Hi,
I did two changes: DIO0 =4 and the pins 22 to 24 jumper.
But I still have issues when running test_lorapy or (lora_util.py) on the RPI3.
I put some print in the code to get it, but I do not understand why I get an "unknown dio3 mapping!"
Do you any hint ?

Thanks,
Ludovic
```
pi@raspberrypi:~/pysx/pySX127x $ python lora_util.py
/home/pi/pysx/pySX127x/SX127x/board_config.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(BOARD.LED, GPIO.OUT)
((' self.dio_mapping[3]: ', None)'selfmode',
1Traceback (most recent call last):
  File "/home/pi/pysx/pySX127x/SX127x/LoRa.py", line 196, in _dio3
)
    raise RuntimeError("unknown dio3 mapping!")
('selfmode', 3)
RuntimeError: Traceback (most recent call last):
unknown dio3 mapping!
  File "lora_util.py", line 35, in <module>
    lora = LoRa(verbose=False)
  File "/home/pi/pysx/pySX127x/SX127x/LoRa.py", line 100, in __init__
    self.rx_chain_calibration(calibration_freq)
  File "/home/pi/pysx/pySX127x/SX127x/LoRa.py", line 858, in rx_chain_calibration
    self.set_freq(freq_bkup)
  File "/home/pi/pysx/pySX127x/SX127x/LoRa.py", line 284, in set_freq
    assert self.mode == MODE.SLEEP or self.mode == MODE.STDBY or self.mode == MODE.FSK_STDBY
(' self.dio_mapping[3]: ', None)
AssertionError
```

mayeranalytics, 2017-12-07:
I suspect you changed the dio_mapping and entered the Raspberry pin number. 

(I’m not quite sure what “and the pins 22 to 24 jumper” means – I take it that you use different pins on the Raspberry.)

Background: The SX127x have 6 DIOs that you can assign a function to each. DIO0, for example, can output RxDone (0), TxDone (1) or CadDone (2). These 0,1,2 values have nothing to do with Raspberry pin numbers, they just identify SX127x functionality. A value of 4 would be illegal (and actually impossible to set). If you want to know more, look at the datasheet, starting with table 18. on p.46.

Instead, change the pin numbers in board_config.py.

gabriel-blacksheep, 2017-12-20:
Hi, I had the same issue, indeed you need to set DIO0 = 4 but make sure you change SWITCH to something else (I used 21) and change DIO3 (which is NC on the Dragino shield) to something toher than 25 or you will conflict with the SPI CS pin (since you have now soldered board-pins 22 & 24 together (which are BCM pins 25 & 8).  Make sure to reboot as well as your GPIO setup will be all weird, but now I have it working.  @mayeranalytics what are your thought on making a Dragino subclass with the correct pins that would allow this to work out of the box?  If someone also knows to to software re-map GPIO to avoid the soldering step this could also be nice.

mcmayer, 2017-12-20:
@artificialsheep Thanks for the input. I don't have a Dragino shield, so I can't test anything. But if you have figured it out, and if you have the time, why don't you just make a pull request?

mayeranalytics, 2017-12-20:
@artificialsheep Thanks for the input. I don't have a Dragino shield, so I can't test anything. But if you have figured it out, and if you have the time, why don't you just make a pull request?

mkshrps, 2018-04-17:
Raspberry Pi stretch has an overlay for this included. To remap the SPI CE0 pin to work with the Dragino add the following to the /boot/config.txt file
dtoverlay=spi0-cs,cs0_pin=25
where 25 is the BCM designation for the GPIO (physical) pin 22. 
you can also remap the SPI CE1 pin if you ever need to e.g.
dtoverlay=spi0-cs,s0_pin=25, cs1_pin=xx
The pin definitions for GPIO, BCM and wiring PI definitions can be found at https://pinout.xyz/

Bakhoj, 2018-05-03:
Hi, 

I'm also trying to use this library with a Dragino Lora GPS Hat v1.4 board on a RPi3, which has to retrieve some data from a RN2483 on a Waspmote Pro v1.5 and is giving me some problems aswell.

I've also soldered the BCM 25 & 8 together and connected the DIO3 to BCM1 since it isn't connected on the board as seen here: [PDF](https://github.com/dragino/Lora/blob/master/Lora_GPS%20HAT/v1.4/Lora%20GPS%20%20HAT%20for%20RPi%20v1.4.pdf) 

At the moment I can recieve data, but only something like once a few minutes if it's sending every 2 second. Only the first 3 bytes have the correct values most of the time while the rest is partly correct, and sometimes it just gets a dump of data.

The code I'm using is a remake of the rx_cont.py:
<details>
<summary>main.py</summary>

```Python
import time
import sys
from SX127x.LoRa import *
from SX127x.board_config import BOARD

print("Start Master Module")

BOARD.DIO0 = 4
BOARD.DIO3 = 1
BOARD.SWITCH = 21
BOARD.setup()

class LoRaMaster(LoRa):
	def __init(self):
		super(LoRaRcvCont, self).__init__(verbose)
		self.set_mode(MODE.STDBY)
		#self.set_dio_mapping([0] * 6) 

	def on_rx_done(self):
		BOARD.led_on()
		print("\nRxDone")
		self.clear_irq_flags(RxDone=1)
		payload = self.read_payload(nocheck=True)
		print(bytes(payload).hex())
		self.set_mode(MODE.SLEEP)
		self.reset_ptr_rx()
		BOARD.led_off()
		self.set_mode(MODE.RXCONT)

	def on_txdone(self):
		print("\nTxDone")
		print(self.get_irq_flags())

	def on_cad_done(self):
		print("\non_CadDone")
		print(self.get_irq_flags())

	def on_rx_timeout(self):
		print("\non_RxTimeout")
		print(self.get_irq_flags())
		time.sleep(.5)
		self.set_mode(MODE.SLEEP)
		self.reset_ptr_rx()
		self.set_mode(MODE.RXCONT)

	def on_valid_header(self):
		print("\non_ValidHeader")
		print(self.get_irq_flags())

	def on_payload_crc_error(self):
		print("\non_PayloadCrcError")
		print(self.get_irq_flags())

	def on_fhss_change_channel(self):
		print("\non_Fhss_changeChannel")
		print(self.get_irq_flags())

[... 187 lines cut ...]
```
pi@raspberrypi:/usr/lib/python3.5/pySX127x $ python3 test_lora.py
F.....
======================================================================
FAIL: test_mode (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_lora.py", line 43, in wrapper
    func(self)
  File "test_lora.py", line 62, in test_mode
    self.assertEqual(lora.get_mode(), m)
AssertionError: 128 != 129

----------------------------------------------------------------------
Ran 6 tests in 0.012s

FAILED (failures=1)
```

</details>

Anyone have an idea what is missing?

Thanks,
Bakhoj

Bakhoj, 2018-05-05:
I fixed my problem with the partly incorrect data from the RN2483.

The Dragino LoRa GPS hat v.1.4's schematics referes to the LoRa module as "RFM92_93W" and I could not find out if it was using RFM92W or RFM93W. The main difference between the 2 is the spreading factor range, which for 92 is 6-12 and the 93 it is 6-9.
After changing the setup to use SF9 instead of 12 it worked.
The error with **lora_util.py** is gone aswell but the same error is still present when running **test_lora.py** is it something to be worried about? I understand it's something with the module being in the wrong mode?

pyflier, 2018-08-22:
Hi Bakhoj
I'm facing exactly the same issue.  but even more over two dio0 and dio1:
File "/home/pi/pySX127x-master/SX127x/LoRa.py", line 164, in _dio0
    raise RuntimeError("unknown dio0mapping!")
RuntimeError: unknown dio0mapping!
Traceback (most recent call last):
  File "/home/pi/pySX127x-master/SX127x/LoRa.py", line 177, in _dio1
    raise RuntimeError("unknown dio1mapping!")

Just wondering if you managed to overcome the issue

MikeRouten, 2018-11-15:
I am trying to get the Dragino board running on a Pi 3 but keep getting the following error:

``` GPIO.setup(BOARD.LED, GPIO.OUT)
Traceback (most recent call last):
  File "tx_ttn.py", line 40, in <module>
    lora = LoRaWANsend(False)
  File "tx_ttn.py", line 15, in __init__
    super(LoRaWANsend, self).__init__(verbose)
  File "/home/pi/LoRaWAN/SX127x/LoRa.py", line 100, in __init__
    self.rx_chain_calibration(calibration_freq)
  File "/home/pi/LoRaWAN/SX127x/LoRa.py", line 856, in rx_chain_calibration
    self.set_freq(freq_bkup)
  File "/home/pi/LoRaWAN/SX127x/LoRa.py", line 282, in set_freq
    assert self.mode == MODE.SLEEP or self.mode == MODE.STDBY or self.mode == MODE.FSK_STDBY
AssertionError
```

I have applied the CS remap as suggested by mkshrps and I have modified the LoRa.py file DI0=4 value.

Does anyone have any suggestions on how to resolve this issue?

mkshrps, 2018-11-19:
Ideally you should use an oscilloscope to check what the pins are actually doing in particular the CS pin. If you can write a loop which reads a register you can then check the cs, Mosi,miso and sclk pins are doing something. This would confirm the remapping  dtoverlay=spi0-cs,cs0_pin=25 is working .  Also make sure the Reset pin is set high.

MikeRouten, 2018-12-05:
For anyone that is interested, I got this working with the following configuration on a fresh install of Raspbian Stretch:
1. Implement the /boot/config.txt changes that mkshrps suggested by adding "dtoverlay=spi0-cs,cs0_pin=25" to the config.txt record.
2. Use the following values in board_config
    DIO0 = 4
    DIO1 = 23
    DIO2 = 24
    DIO3 = 21
    LED  = 18
    SWITCH = 7
3. ADD THE FOLLOWING LINES TO constants.py
@add_lookup
class SPI_BAUD_RATE:
    MAX_SPEED_HZ = 5000
@add_lookup
class SPI_MODE:
    SPI_MODE = 0b01

4. ADD THE FOLLOWING LINES TO LoRa.py at the top of the class definition right after "    spi = BOARD.SpiDev() ":
    spi.max_speed_hz = SPI_BAUD_RATE.MAX_SPEED_HZ
    spi.mode = SPI_MODE.SPI_MODE

With these changes / additions I was able to get the Dragino Hats talking to each other over the Pi's without having to do any soldering on the Hat.

xmrhahnx, 2019-03-29:
HI there, i'm facing with similar problem. I made the modification as MikeRouten suggested and the test_lora.py works as expected. But when i run the rx_cont.py or any other example i get this:
`/home/pi/pySX127x-master/SX127x/board_config.py:58: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(BOARD.LED, GPIO.OUT)
Traceback (most recent call last):
  File "rx_cont.py", line 101, in <module>
    print(lora)
  File "/home/pi/pySX127x-master/SX127x/LoRa.py", line 945, in __str__
    s += " pa_dac             %s\n" % ['default', 'PA_BOOST'][self.get_pa_dac()]
  File "/home/pi/pySX127x-master/SX127x/LoRa.py", line 56, in wrapper
    return func(self, self.spi.xfer([register_address, 0])[1])
  File "/home/pi/pySX127x-master/SX127x/LoRa.py", line 818, in get_pa_dac
    raise RuntimeError("Bad PA_DAC value %s" % hex(pa_dac))
RuntimeError: Bad PA_DAC value 0x3`

any suggestion?

amotl, 2020-04-28:
Hi there.

### Introduction
We spent some time on revamping these libraries to be used within our [Terkin Datalogger](https://github.com/hiveeyes/terkin-datalogger). While adding some bits from the community and general polishing, the main contribution from our pen was https://github.com/daq-tools/dragino/commit/772f552d, in order to be able to configure the Dragino class at runtime to become independent of the `dragino.ini` file. We hope this is as useful for the community as it was important to us.

### Resources
The trees we have modified:
- https://github.com/daq-tools/pySX127x/commits/dragino
- https://github.com/daq-tools/dragino/commits/terkin

### Usage
The code is driven by [terkin.network.lora.LoRaDriverDragino](https://github.com/hiveeyes/terkin-datalogger/blob/0.8.0/src/lib/terkin/network/lora.py#L176-L216). You might want to use this as an example blueprint.

### Installation
```
mkdir lib

# Install driver support for Dragino LoRa Hat.
curl --location https://github.com/daq-tools/dragino/archive/terkin.tar.gz | tar -C lib --strip-components=1 -xzvf - dragino-terkin/dragino

# Install updated pySX127x driver.
curl --location https://github.com/daq-tools/pySX127x/archive/dragino.tar.gz | tar -C lib/dragino --strip-components=1 -xzvf - pySX127x-dragino/SX127x
```

### Credits
Thanks a bunch to @jeroennijhof, @pjb304 of @computenodes, @mcmayer of @mayeranalytics and @jonashoechst of @Nature40!

Cheers,
Andreas.

chuanisawesome, 2022-05-07:
Hi,

I am new to using LoRa and the Dragino Hat for Raspberry pi. I am currently working on a project that I would like to have the Heltec LoRa32 talk to the Dragino Hat locally. Is there any suggestions on how to get started or if this is a possibility? I've ran the lora_test.py and I get this error:

[image]

Thank you!

mayeranalytics, 2022-05-09:
The failed test tells you that the LoRa chip is not in the correct mode, or registers don't have the expected values. This usually means that there are communication problems. When I look at your log, I see that the read commands seem to always return `0`, and that is quite suspicious.

chuanisawesome, 2022-05-10:
I end up using Heltec LoRa32 as my gateway since I'm having so much trouble with Dragino LoRa Hat for the RPi.

amotl, 2022-05-10:
Hi again,

following up on our last post at https://github.com/mayeranalytics/pySX127x/issues/21#issuecomment-620695583, we would like to bring to your attention that @BNNorman picked up the torch and has been working on his fork [^1] continuously since May 2021.

For further information, see also https://github.com/daq-tools/dragino-lorawan/issues/1 where Brian shared an overview about the improvements and features he has been working on, status November 2021.

With kind regards,
Andreas.

[^1]: https://github.com/BNNorman/dragino-1

/cc @tonkenfo

chuanisawesome, 2022-05-10:
Thank you so much @amotl ! I will look into it. I really appreciate it

BNNorman, 2022-05-12:
> I end up using Heltec LoRa32 as my gateway since I'm having so much trouble with Dragino LoRa Hat for the RPi.

Did you really mean Gateway? Gateways are supposed to be able to receive on 8 channels simultaneously.  Single channel gateways are not supported by TTN (AFAIK).

Check your Dragino LoRA/GPS HAT version is V1.4 - it has DIO signals wired that previous versions do not. That may be your real issue with RPi.