mayeranalytics/pySX127x #23: Unable to run

lakewik, 2017-12-11:
I receiving this error while trying to run lora_util.py
/home/pi/2/pySX127x/SX127x/board_config.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(BOARD.LED, GPIO.OUT)
Traceback (most recent call last):
  File "lora_util.py", line 35, in <module>
    lora = LoRa(verbose=False)
  File "/home/pi/2/pySX127x/SX127x/LoRa.py", line 100, in __init__
    self.rx_chain_calibration(calibration_freq)
  File "/home/pi/2/pySX127x/SX127x/LoRa.py", line 856, in rx_chain_calibration
    self.set_freq(freq_bkup)
  File "/home/pi/2/pySX127x/SX127x/LoRa.py", line 282, in set_freq
    assert self.mode == MODE.SLEEP or self.mode == MODE.STDBY or self.mode == MODE.FSK_STDBY
AssertionError

mcmayer, 2017-12-11:
Can you give more information?
- what SBC, what RF module are you using?
- did you change anything (e.g. board_config.py)
- does SPI communication work?
- anything else unusual?

lakewik, 2017-12-12:
**1.what SBC, what RF module are you using?**

   RaspberryPi 3,  LoRa Ra-01, SX1278

**2. did you change anything (e.g. board_config.py)?**

   I don't changed anything in code

**3. does SPI communication work?**

   Yes

**4. anything else unusual?**

I tested all examples (  lora_util.py,  	rx_cont.py, test_lora.py, tx_beacon.py), but i got same AssertionError error

shashfrankenstien, 2017-12-26:
Same error

awrede, 2018-01-02:
Same error as @lakewik, also on a Raspberry Pi 3 but with the Hope RF96 on a Adafruit breakout board. At the time of the assert, the value of mode is 3.  Is there a chance that this is a RPi 2 vs. RPi 3 issue?

awrede, 2018-01-03:
I got access to a RPi 2 and can confirm that the assert is not triggered using the same radio module.

SloMusti, 2018-01-10:
We have traced this issue to some change in the OS as such, not a bug in the library, see https://github.com/IRNAS/PiRA-zero-firmware/issues/15#issuecomment-356692020

Also does not work on RPi 3 and Raspbian Stretch Lite 2017-11-29

mayeranalytics, 2018-01-11:
@SloMusti Thanks for uncovering this! We'll get a Rasp3 and have a look.

Cyllos42, 2018-01-30:
I can confirm this error also appears on a Pi Zero W with Raspbian Stretch Lite 2017-11-29...

JacquesLG, 2018-01-30:
Hello,
is there any update on this issue?
I really need the library for a school project 😬 

Thanks

SloMusti, 2018-01-30:
@JacquesLG sinmply use a sufficiently old raspbian, say Jesse or so.

mayeranalytics, 2018-01-31:
@JacquesLG get the newest version and try again (SPI clock speed is now limited to 5MHz)

mayeranalytics, 2018-02-01:
**Explanation:**
For some reason the spidev max clock speed defaults to 12MHz now instead of 10MHz previously. (If you look at the SPI speeds that the BCM283x processors actually support, i.e. 125MHz/(2n+1) according to this here, neither of these numbers makes any sense.)
And 10MHz happens to be the max clock speed that the SX127x can support.
RPi internals are obscure (try to find the processor datasheet!), so within the given limitations this is a sensible fix.

closed 2018-02-06 by mayeranalytics

Cyllos42, 2018-03-06:
Using the newer version doesn't change a bit :(
I have a Pi Zero W with the latest pySX127x (commit d489638)..
Have connected a 433MHz RFM96 module to the pins of the Pi as suggested in the README.md.
Using a print statement
>#LoRa.py line 282
> print(self.mode, MODE.SLEEP, MODE.STDBY, MODE.FSK_STDBY)
 it prints 1, 0 and it crashes.

bkrajendra, 2019-12-04:
Same with RPIb3+ with Raspbian Buster .

```
/home/pi/lora/pySX127x/SX127x/board_config.py:61: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(BOARD.LED, GPIO.OUT)
/home/pi/lora/pySX127x/SX127x/board_config.py:62: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(BOARD.RST, GPIO.OUT)
Traceback (most recent call last):
  File "./LORA_SERVER.py", line 115, in <module>
    assert(lora.get_agc_auto_on() == 1)
AssertionError
```

djthegr8, 2021-04-18:
@mayeranalytics could you please update on this?

mayeranalytics, 2021-04-19:
I don't think this has to do with the library, but rather with the SPI communication. Did you try to reduce the spidev clock speed?

djthegr8, 2021-04-19:
The problem turned out to be that GPIO4 (pin 7) also works as GPCLK0.

kapaA, 2022-01-20:
any update solution on this issue?

mayeranalytics, 2022-01-21:
The assertions check whether the chip is in the state it should be according to the commands sent. Usually, when you get this, there is a communication problem. I cannot debug your communications problems for you. What have you tried so far?

mayeranalytics, 2022-01-21:
> any update solution on this issue?

The issue is closed, so what update do you expect?

Again: The assertions check whether the chip is in the state it should be according to the commands sent. Usually, when you get this, there is a communication problem. 

I cannot debug your communications problems for you. What have you tried so far?

spinedaospina, 2022-07-22:
I've been working in this issue, maybe the problem is in the 0x01 byte in the register because working in 433MHz is a Low Frequency for the module (Modifying the third bit), due to a lot of logic is setted for High freq (>815MHz) we get a lot of asserts... 
At the moment I still stuck, but maybe in the future this comment helps someone.