mayeranalytics/pySX127x #31: SX128x compatibility ?

kojakian, 2019-06-20:
Hi everyone,

I'm trying to work on the SX1280 with python for a prototype, do you think this interface can be compatible or could it be a good base for a python interface ?

If you have info on an existing project trying to achieve that, that would be very helpful, because I can't find anything !

Thanks for your help

mayeranalytics, 2019-06-22:
Out of the box the pySX127x certainly won't work. And the interface of the 1280 looks very different from the 127x one so you have to start from scratch, I think.
Maybe this here, [chisl.io](https://chisl.io), can help you with that.

maholli, 2019-08-01:
Just happened to see this issue. My [CircuitPython_SX1280 library](https://github.com/maholli/CircuitPython_SX1280) is still a work in progress, but it's confirmed working for LoRa. It's a mess but might give you somewhere to start.

kojakian, 2019-08-01:
Hi ! Thank you both for your help :) 

Right now I'm working from the original code https://os.mbed.com/teams/Semtech/code/SX1280DevKit/ 
Which is in C++ ! 
The final code will be pushed on this device : https://wireless-solutions.de/products/radiomodules/im282a.html

Right now it's better for me to go this way ! @maholli I was already following your project and I will continue to see where it goes ! I had a question for you though but couldn't find a way to contact you through github so I'm glad you're writing me there ! I was wondering why did you choose to make your own circuit for the SX1280, when you could have used a device such as the E28-2G4M20S (https://fr.aliexpress.com/item/32865262236.html) ? Is it a personal choice ?

Also, are you using a nucleo to communicate with your circuit ?

maholli, 2019-08-01:
Hi, no problem!

I made my own breakout board the for SX1280 because I wanted one with the same footprint as many other LoRa radios (RF9x, RF69) so it could be interchangeable. 

I'm using my own development board(s) to communicate with the radio. See https://github.com/maholli/SAM32

kojakian, 2019-08-01:
Very interesting, I will look into that thanks :)

closed 2022-09-23 by mayeranalytics