mayeranalytics/pySX127x #32: Error when running test_lora.py

Oseihie, 2019-06-29:
Hi there is already an issue with a similar title but my error is different.
I am using rpi 3 and i have triple checked that my connections are right and soldering is okay. 
I get this error when i run the python file:

> ======================================================================
FAIL: test_mode (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./test_lora.py", line 43, in wrapper
    func(self)
  File "./test_lora.py", line 62, in test_mode
    self.assertEqual(lora.get_mode(), m)
AssertionError: 128 != 129

----------------------------------------------------------------------
Ran 6 tests in 0.011s

FAILED (failures=1)

Any help will be appreciated.

mayeranalytics, 2019-07-09:
These types of error usually occur when the SPI communication is faulty. The error basically says that the LoRa chip is not in the mode it should be in. Try reducing the SPI clock speed.

Oseihie, 2019-07-11:
thank you for your reply. i changed the clock speed to 1Mhz, but i got this error.
i changed it back to what it was before and i still got this errior below.

>  
GPIO.setup(BOARD.RST, GPIO.OUT)
FFFFFF
======================================================================
FAIL: test_mode (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_lora.py", line 43, in wrapper
    func(self)
  File "test_lora.py", line 62, in test_mode
    self.assertEqual(lora.get_mode(), m)
AssertionError: 8 != 129

======================================================================
FAIL: test_set_agc_on (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_lora.py", line 43, in wrapper
    func(self)
  File "test_lora.py", line 74, in test_set_agc_on
    self.assertEqual((get_reg(REG.LORA.MODEM_CONFIG_3) & 0b100) >> 2, 1)
AssertionError: 0 != 1

======================================================================
FAIL: test_set_dio_mapping (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_lora.py", line 43, in wrapper
    func(self)
  File "test_lora.py", line 90, in test_set_dio_mapping
    self.assertEqual(get_reg(REG.LORA.DIO_MAPPING_1), 0b01010101)
AssertionError: 0 != 85

======================================================================
FAIL: test_set_freq (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_lora.py", line 43, in wrapper
    func(self)
  File "test_lora.py", line 68, in test_set_freq
    lora.set_freq(f)
  File "/home/pi/pySX127x/SX127x/LoRa.py", line 283, in set_freq
    assert self.mode == MODE.SLEEP or self.mode == MODE.STDBY or self.mode == MODE.FSK_STDBY
AssertionError

======================================================================
FAIL: test_set_low_data_rate_optim (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_lora.py", line 43, in wrapper
    func(self)
  File "test_lora.py", line 81, in test_set_low_data_rate_optim
    self.assertEqual((get_reg(REG.LORA.MODEM_CONFIG_3) & 0b1000) >> 3, 1)
AssertionError: 0 != 1

======================================================================
FAIL: test_setter_getter (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_lora.py", line 55, in test_setter_getter
    self.assertEqual(lora.get_payload_length(), l)
AssertionError: 0 != 1

----------------------------------------------------------------------
Ran 6 tests in 0.018s

FAILED (failures=6)

please any suggestion?

mayeranalytics, 2019-07-12:
it could be any number of things and it's difficult to diagnose remotely. Wiring could be faulty, communication could be faulty, the chip could be faulty. Maybe even the rasp could be faulty (I never heard of such a case, though).
SPI has some parameters (clock polarity and phase, etc. See [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md)). Maybe those are somehow messed up.

In the end, I think, you have to go back to basics and establish whether you can successfully communicate with the chip at all. I.e. forget the library for the moment and just try to read a register, for example RegOpMode (default 0x01) or RegOcp (default 0x2B). It's quite easy to do with python's `spidev`.

And when the software fiddling fails, you eventually have to attach a scope of logic analyzer to figure out what's truly going on.

Oseihie, 2019-07-14:
> it could be any number of things and it's difficult to diagnose remotely. Wiring could be faulty, communication could be faulty, the chip could be faulty. Maybe even the rasp could be faulty (I [...]

Ok. i am sure its not a problem with the Lora Chip because arduino to arduino communication works.
So i will use an oscilloscope to figure out the problem

Thank you :)

HarishMohanty, 2019-07-20:
Hi@Oseihie

Did you find any solution to this issue ?
I am having the same issue.All 6 test cases failed.
Please post if you find any solutions.

> Ok. i am sure its not a problem with the Lora Chip because arduino to arduino communication works.
> So i will use an oscilloscope to figure out the problem
>
> Thank you :)

mayeranalytics, 2019-07-22:
What is your setup (which SBC, which LoRa module, etc.)?

HarishMohanty, 2019-07-27:
> 
> 
> What is your setup (which SBC, which LoRa module, etc.)?

Hi sir
I am using Lora Sx1278 Module with raspberry pi 3B + and Raspbian Buster OS.

Oseihie, 2019-07-29:
> Hi@Oseihie
>
> Did you find any solution to this issue ?
> I am having the same issue.All 6 test cases failed.
> Please post if you find any solutions.

Hey,
no i havent figured out the problem because i have left the project for a while now. If i do figure it out, i will let you know.

FredyOngJaya, 2020-01-21:
> Hi there is already an issue with a similar title but my error is different.
> I am using rpi 3 and i have triple checked that my connections are right and soldering is okay.
> I get this error when i [...]

I found solution for this test case. It need to add delay between write and read (1ms is good enough to pass the test).

closed 2020-01-21 by Oseihie

the-asymptote15, 2022-06-17:
> > Hi there is already an issue with a similar title but my error is different.
> > I am using rpi 3 and i have triple checked that my connections are right and soldering is okay.
> > I get this error when i run the python file:
> > > ======================================================================
> > > FAIL: test_mode (**main**.TestSX127x)
> > 
> > 
> > Traceback (most recent call last):
> > File "./test_lora.py", line 43, in wrapper
> > func(self)
> > File "./test_lora.py", line 62, in test_mode
> > self.assertEqual(lora.get_mode(), m)
> > AssertionError: 128 != 129
> > Ran 6 tests in 0.011s
> > FAILED (failures=1)
> > Any help will be appreciated.
> 
> I found solution for this test case. You need to add delay between write and read (1ms is good enough to pass the test).

Can you elaborate more on how we can do that?