mayeranalytics/pySX127x #34: Calculating SNR

niclas0219, 2019-09-27:
I am wondering if the calculation of the SNR is correct. The datasheet (sx1278) says the following: 

> Estimation of SNR on last packet received.In two’s compliment format multiplied by 4
SNR dB = PacketSnr [twos complement] / 4

This is achieved with the following code:
`    def get_pkt_snr_value(self):
        v = self.spi.xfer([REG.LORA.PKT_SNR_VALUE, 0])[1]
        return float(256-v) / 4.`

This is a bit over my head but when i look at the RAW value that i get from the register i see "0b101100" = decimal 44.
256 - 44 = 212
212 / 4 = 53.

What does the 256 - v do?

niclas0219, 2019-09-28:
Found some code on this site: [https://blog.michaelyin.info/convert-8bit-byte-to-signed-int/]

if v > 127:
    return (256-v) * (-1)
else:
    return v

This would seem to give a better result considering the SNR in the SX1278 register is not always negative.

mayeranalytics, 2019-10-01:
You are absolutley right, the `if v>127 ...` is missing.

closed 2019-10-01 by mayeranalytics