mayeranalytics/pySX127x #36: AssertionError

Stefan213, 2020-10-23:
My project has worked 4 months ago, but i haven't used it since. I am now using a PCB and when i execute the test_lora.py programm i get following error:  

======================================================================
FAIL: test_mode (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "test_lora.py", line 43, in wrapper
    func(self)
  File "test_lora.py", line 62, in test_mode
    self.assertEqual(lora.get_mode(), m)
AssertionError: 128 != 129

----------------------------------------------------------------------

Thank you in advance.

mayeranalytics, 2020-10-23:
The assertion tests whether the chip is indeed in the state it should be in. When that is not the case usually there is a SPI communication problem.

Stefan213, 2020-10-23:
Do you know how to pin down the problem?

mayeranalytics, 2020-10-23:
Back to basics: Directly use Wiring-Pi (I assume it's a rasp) and write, then read, some register, e.g. the mode register.

Stefan213, 2020-11-06:
I tested this and everything works. I tested the board with an Arduino and it worked perfectly fine.

Stefan213, 2020-11-10:
If i run rx_cont there are no errors but i dont receive any packets.

mayeranalytics, 2020-11-10:
I don't really know how to debug for you this remotely. I can only repeat what I already said: Ignore the library for the moment and try to establish communication on the Raspberry using the simplest setup possible.

Stefan213, 2020-11-10:
I have already done this. I don't know where the problem lies.

mayeranalytics, 2020-11-11:
Have they maybe brought out a new chip version? (What Raspberry are you using anyway? I'm sitting on a collection of Rasp 2s and 3s. Rasp 4 isn't that appealing to me so I don't even have one.)

Stefan213, 2020-11-11:
It doesn't seem like there is a new version. I am using a Raspberry 3B+.

oliverpaadik, 2020-11-11:
I actually ran into similar issue the other day. Thought that the module itself is faulty, but connecting a new one to RPi and running the test, I got the same assertion error. However, this doesn't cause any issues with creating a communication link between two modules -- so in your case, maybe the problem lies somewhere else as well?

Stefan213, 2020-11-17:
I finally found the error. I was testing another transceiver and it worked, so i took a closer look on the other one and the crystal oscillator was missing. I wonder why it was missing. 

Thank you for your support, i am very happy that i finally can close this issue.

closed 2020-11-17 by Stefan213