mayeranalytics/pySX127x #37: Tried to send data from RPi to Arduino but Arduino did not receive data

paulcastor30, 2021-04-05:
Hello, i am trying to perform the client and the server side of which arduino is the client and RPi is the server based on the rpsreal/LoRa_Ra-02_Arduino given examples. However, when RPi send data to Arduino, Arduino does not receive data from the RPi.
I have changed nothing from the code. I use sx1278 Ra-02 module.

mayeranalytics, 2021-04-06:
Are you using the rpsreal fork of this library? If yes, then you probably should ask rpsreal, because I don’t really know what this is, exactly. If no, you should probably try the rpsreal fork first.

Reply to: mayeranalytics/pySX127x ***@***.***>
Date: Monday, 5 April 2021 at 19:58
To: mayeranalytics/pySX127x ***@***.***>
Cc: Subscribed ***@***.***>
Subject: [mayeranalytics/pySX127x] Tried to send data from RPi to Arduino but Arduino did not receive data (#37)

paulcastor30, 2021-04-06:
Thank you very much for your response. I used the pySX127x library and the rx_cont.py example for my RPi with sx1278 Ra-02 module  and successfully received data from my arduino with sx1278 Ra-02 module. However, when i try the other way around and use the tx_cont.py to send data to my arduino, i cannot see any received data in my arduino. 

I already tried rps fork of this library but i cannot get it to work.

atonughosh, 2021-04-14:
I am also having the same issue as @paulcastor30 
Someone, please help with this.

thanapol199717, 2021-04-18:
I am also having the same issue as @paulcastor30

I use SX1276 module.

I need help. Because i've been searching for many days I want to cry.

paulcastor30, 2021-04-24:
@thanapol199717  @atonughosh did you succeeded in making the library work for the RPi to send data and for arduino to receive the data?

thanapol199717, 2021-04-24:
Now I can send data to the arduino using Raspi, I use this one. 
https://github.com/raspberrypi-tw/lora-sx1276/blob/master/02-p2p/p2p_send.py >> For Raspi
https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiver/LoRaReceiver.ino >> For Arduino

Right now, I still can't get it back and forth.

paulcastor30, 2021-04-24:
@thanapol199717 thank you very much for your reply. 

I can now send data from RPi to Arduino and vice versa. This are the libraries I used to make it work.
[http://www.airspayce.com/mikem/arduino/RadioHead/RadioHead-1.116.zip](url) >> for Arduino
[https://github.com/adafruit/Adafruit_CircuitPython_RFM9x](url) >> for RPi

Just experiment with the examples given there. It works with the RFM95 and/or RFM95W module only.

thanapol199717, 2021-04-24:
@paulcastor30 thank you very much
Can you tell me which code you use, because RFM95 is not used? I want it to communicate with each other to control some devices.

paulcastor30, 2021-04-24:
@thanapol199717 may i know what specific module did you use on your arduino and RPi?

I already got my sx1278 module working for both my Arduino and RPi to send data vice versa continously.

paulcastor30, 2021-04-24:
@thanapol199717 I used the following:

[https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/blob/master/examples/rfm9x_rpi_simpletest.py](url) >> RPi server. Send data to arduino with sx1278 module and wait for a response from arduino at a certain time interval.
**RadioHead\examples\rf95\rf95_server.pde** >> Arduino with sx1278. Receive a data from RPi and send a datain return as a response.

thanapol199717, 2021-04-24:
@paulcastor30 I am using the SX1276 module, I want Raspi and Arduino to send data vice versa.

I still can't find a way, can you tell me? I don't know if the example you provided is with my SX1276 module, because I am using the Lora.h library at Arduino.

paulcastor30, 2021-04-24:
@thanapol199717 do you have any means online so i can contact you? i am willing to help you.

thanapol199717, 2021-04-24:
@paulcastor30  I have an email thanapol199717@gmail.com. And this is my facebook Thanapol pornanake. Or do you have other channels? So I can apply to contact you. Thank you very much Paulcastor30

paulcastor30, 2021-04-24:
Okay sure, i am gonna send an email to help you. thanks.

thanapol199717, 2021-04-24:
Thank you very much

Syrex-o, 2021-06-29:
@paulcastor30 are you open to drop me a mail as well? 
I can send data from:
- PI to to PI 
- Arduino to PI. 

But not from Pi to Arduino with callbacks. 

alberthenkel.ah@gmail.com
Thank you very much.

chuanisawesome, 2022-05-07:
Hi,

I was wonder where can I find the solution to send data from Arduino to RaspberryPi? I need help with my project.

Thank you in advance!

mayeranalytics, 2022-05-09:
Arduino to Pi should work, according to the comments here. If you tell us what hardware & library you are using then maybe someone can help.

Beerocket7, 2023-08-29:
@paulcastor30 can you drop me a mail about your setup?
I've tried to send data from RPi to ESP32 without receive any data

Thank you for your support 

alessandroBee@protonmail.com

Beerocket7, 2023-08-29:
Quick update, resolve by using the library [Adafruit_CircuitPython_RFM9x](https://github.com/adafruit/Adafruit_CircuitPython_RFM9x)
Rock you guys