mayeranalytics/pySX127x #40: sx1278 reciving problem

ShootGan, 2021-11-11:
Hello,
I want to use sx1278 with raspberry pi 4 to receive data from Arduino with the same module. I don't get any message and when I run test_lora.py i get one test fails. 
`F.....
======================================================================
FAIL: test_mode (__main__.TestSX127x)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pi/pySX127x/test_lora.py", line 43, in wrapper
    func(self)
  File "/home/pi/pySX127x/test_lora.py", line 62, in test_mode
    self.assertEqual(lora.get_mode(), m)
AssertionError: 128 != 129

----------------------------------------------------------------------
Ran 6 tests in 0.006s

FAILED (failures=1)
`
I don't really know with what i have problem. Can someone help me? Thank you in advance.

liuhengjing-sudo, 2021-12-10:
Hello 
I have the same problem but mine is 0!=129
my module is lora shield v1.4 and rasp pi 3
Have you address the problem?
Thanks

lalmiAhmed, 2022-12-08:
I don't know what is the exact source of the problem, but this can be solved if you just set a lora to stdby mode before the loop:
lora.set_mode(MODE.STDBY)