mayeranalytics/pySX127x #42: Question about link rates and multicast

pablogranolabar, 2022-07-05:
I have a unique use case that I am considering pySX127x for and I am curious if this would be a possible configuration. I have a large number of sensors which receive information in a one-way serial fashion, using a UART speed of 115.2K. These are broadcast messages that go to all end nodes and without a requirement for two way communications. Is this a possibility with the Semtech line and pySX127x?

Also I can't make heads or tails out of the Semtech datasheet and data calculator table for coding rates, any ideas what an appropriate coding rate for 115.2K would be?

mayeranalytics, 2022-07-06:
Have a play with one of the LoRa bitrate calculators, e.g. [this one](https://unsigned.io/understanding-lora-parameters/). You'll see that 115.2Kbit/s isn't achievable with LoRa. And keep in mind that there is a tradeoff between range and bitrate, and that at high bitrates you may have to consider latency, too. 
By the way, you can reduce the size of the payload significantly by using clever encoding/preprocessing/compression of the data - 115.2kbit/s from a sensor sounds like a lot to me.

pablogranolabar, 2022-07-07:
These 2.4Ghz modules are touting multi megabit speeds now at 500Khz I thought?

Agreed on compression, however the problem there is a requirement for intelligence on the sensor side.

The data is in the opposite direction actually, a bunch of sensors get constant 115.2K serial updates but no requirement for the sensor to communicate back with the base station.

pablogranolabar, 2022-07-29:
This is for GPS <1cm accuracy based upon GPS satellite signal degradation due to atmospheric . A site survey is done at a specific locale, and collects measurements which are then used to provide correctional accuracy information that is broadcast to multiple GPS units (up to 500), running onboard a semi-autonomous drone. For precise swarm control over drones in-flight.

pablogranolabar, 2022-08-10:
Some of these 2.4GHz LoRa modules are pushing 500mw so I think the power should be sufficient. But I'm trying to figure out the best way of broadcast on the same channel, some method of multicast or broadcast perhaps via assigning the same LoRa receiver IDs? All endpoints receive the same data so maybe I can do this at the link layer.