mayeranalytics/pySX127x #43: Converting the payload or data to Float

nickyandre, 2022-11-14:
Is that possible to convert the payload received in the Raspberry Pi to Float, Int, or String if yes how to do it? Could you please give me some idea how to do it. Thank you very much.

mayeranalytics, 2022-11-14:
Ok that's a very broad question. The payload is binary, it can be anything you like, or whatever the sender specifies. 
In any case, you have to decode the payload, e.g. with Python [struct](https://docs.python.org/3/library/struct.html). But there are all kinds of binary encodings.

nickyandre, 2022-11-15:
In this case i am sending accelerometer sensor data from microcontroller to Raspberry Pi. The Pi successfully received the data in the payload then decoded with utf8 and converted into variable named data according to the code. In this case we can encode the data using something else instead of utf8? like what you mention using struct?. The sensor is already processed by microcontroller in this case ESP32.

mayeranalytics, 2022-11-16:
Utf8 would be very inefficient, unless you have pure text data. Have a look at [msgpack](https://msgpack.org/), it supports many languages, you can use C to encode on the ESP32 and python on the Rasp to decode.

closed 2022-11-23 by mayeranalytics